gtkwave/gtkwave #438: High memory when viewing VCD files with gtkwave

IceMilkTea, 2025-05-28:
Dear  Sir/Madam:
I have an experience issue regarding viewing VCD files with gtkwave .
 When opening a VCD file exceeding 500MB with gtkwave, the actual memory usage for waveform display can reach 3GB or more.
I have tried using the -o parameter in gtkwave, but the memory usage remains high. I would like to ask if there is a better version that addresses the excessive memory consumption issue to prevent the computer from freezing?
Looking forward to your reply. Thank you very much.

tbybell, 2025-05-28:
The -o command line argument is going to call vcd2fst so there will be some extra time involved for conversion to FST.

If you already have an FST file, initialization should be fairly quick unless you have a large number of signals (I'm talking in the millions of signal range).  If it's 3GB and there are no waves drawn on screen, that tells me that you might have some very deep hierarchies, ones with long names, or many signals.  If there are waves on screen, high memory usage can also indicate a huge number of clock cycles/transitions for signals being viewed.

If your simulator can natively emit FST, I'd suggest having it do that to cut down on conversion time.  If you're using VCS, have it emit FSDB and recompile with the --enable-fsdb option (after pointing $FSDBREADER_HDRS and $FSDBREADER_LIBS to the appropriate directories).

tbybell, 2025-05-29:
How many signals are in your model?  How many clock ticks?

For example:
```
525:/tmp> cat Vexample.vcd | grep -c '^#'
431823
526:/tmp> cat Vexample.vcd | grep -c '^\$var'
57312
```

It sounds to me like your VCD might not contain what is typical VCD usage data.  Is it a low number of signals with a very large number of transitions (count in the millions or tens of millions) on those signals?  If so, there's nothing that can be done.

Please also report what version 3.3.124 uses if you can, as I am curious if any memory allocation issues have turned up in the development branch.  Note that memory usage as reported by /bin/top and other system tools is ridiculously large for any process due to common memory regions spanning across dozens of shared libraries.  Bringing up gtkwave without a file on my machine shows 600MB, which obviously is not due to malloc().

closed 2025-05-29 by IceMilkTea