gtkwave/gtkwave #440: Issues with jumping to time with Logfile viewer

chrisgelinek-rl, 2025-06-12:
Thanks for GTKWave!

I just generated a logfile using [Icarus Verilog](https://steveicarus.github.io/iverilog/), using the built-in `$info` system function call. Here's the code:

```
`timescale 1ns / 1ps
module vt ();
  reg s;
  initial begin
    s = 0;
    #1 s = 1;
    #1 s = 0;
    #0.250 s = 1;
    #0.250 s = 0;
    #0.250 s = 1;
    #0.250 s = 0;
    #0.250 s = 1;
    #0.250 s = 0;
    $timeformat(-9,3,"ns", 0);
    #0.250 s = 1;
    #0.250 s = 0;
    #0.250 s = 1;
    #0.250 s = 0;
    #0.250 s = 1;
    #0.250 s = 0;
  end
  
  always @ (s) begin
    $info("s -> %h at $time(d)=%0d, $time(t)=%0t, $realtime(f): %0f, $realtime(t): %0t why's that?", s, $time, $time, $realtime, $realtime);
  end
endmodule
```

Apart from the usual Verilog weirdness about the behaviour of its time-related system functions as you can see in the log below, `$info` also outputs an extra line with information about the current time and scope in the format `Time: <value> Scope: <scope>`.

I was hoping to leverage log time values in GTKWave's "Log viewer" to jump the view and cursor to the time of the message, but there are a few issues:

 1. When double-clicking on a the integer after `Time: `, the view doesn't jump to the specified time. Interestingly, double-clicking on the same integer number after `$time(t)=` and `$realtime(t): ` works. This looks like a bug to me.
 2. Selecting floating-point values like `3.750ns` also cause a jump to the specified time, unfortunately it's not as convenient as one has to carefully select the whole number including units as double-click doesn't select past the decimal point. I see this as a usability/convenience issue.

Here is the full log file:

```
INFO: vt.v:25: s -> 0 at $time(d)=0, $time(t)=0, $realtime(f): 0.000000, $realtime(t): 0 why's that?
      Time: 0 Scope: vt
INFO: vt.v:25: s -> 1 at $time(d)=1, $time(t)=1000, $realtime(f): 1.000000, $realtime(t): 1000 why's that?
      Time: 1000 Scope: vt
INFO: vt.v:25: s -> 0 at $time(d)=2, $time(t)=2000, $realtime(f): 2.000000, $realtime(t): 2000 why's that?
      Time: 2000 Scope: vt
INFO: vt.v:25: s -> 1 at $time(d)=2, $time(t)=2000, $realtime(f): 2.250000, $realtime(t): 2250 why's that?
      Time: 2250 Scope: vt
INFO: vt.v:25: s -> 0 at $time(d)=3, $time(t)=3000, $realtime(f): 2.500000, $realtime(t): 2500 why's that?
      Time: 2500 Scope: vt
INFO: vt.v:25: s -> 1 at $time(d)=3, $time(t)=3000, $realtime(f): 2.750000, $realtime(t): 2750 why's that?
      Time: 2750 Scope: vt
INFO: vt.v:25: s -> 0 at $time(d)=3, $time(t)=3000, $realtime(f): 3.000000, $realtime(t): 3000 why's that?
      Time: 3000 Scope: vt
INFO: vt.v:25: s -> 1 at $time(d)=3, $time(t)=3000, $realtime(f): 3.250000, $realtime(t): 3250 why's that?
      Time: 3250 Scope: vt
INFO: vt.v:25: s -> 0 at $time(d)=4, $time(t)=4.000ns, $realtime(f): 3.500000, $realtime(t): 3.500ns why's that?
      Time: 3500 Scope: vt
INFO: vt.v:25: s -> 1 at $time(d)=4, $time(t)=4.000ns, $realtime(f): 3.750000, $realtime(t): 3.750ns why's that?
      Time: 3750 Scope: vt
INFO: vt.v:25: s -> 0 at $time(d)=4, $time(t)=4.000ns, $realtime(f): 4.000000, $realtime(t): 4.000ns why's that?
      Time: 4000 Scope: vt
INFO: vt.v:25: s -> 1 at $time(d)=4, $time(t)=4.000ns, $realtime(f): 4.250000, $realtime(t): 4.250ns why's that?
      Time: 4250 Scope: vt
INFO: vt.v:25: s -> 0 at $time(d)=5, $time(t)=5.000ns, $realtime(f): 4.500000, $realtime(t): 4.500ns why's that?
      Time: 4500 Scope: vt
INFO: vt.v:25: s -> 1 at $time(d)=5, $time(t)=5.000ns, $realtime(f): 4.750000, $realtime(t): 4.750ns why's that?
      Time: 4750 Scope: vt
INFO: vt.v:25: s -> 0 at $time(d)=5, $time(t)=5.000ns, $realtime(f): 5.000000, $realtime(t): 5.000ns why's that?
      Time: 5000 Scope: vt
```

tbybell, 2025-06-12:
The whole numbers (or a scaled version) are what is dumped into the VCD or FST file.  I'd guess what's being dumped is in picoseconds because of time timescale 1ns / 1ps.  Look in the VCD to see if the scale for a raw time value isn't 1000x really what you think it is.

chrisgelinek-rl, 2025-06-12:
I'm on Debian 12 (Bookworm) and which provides GTKWave v3.3.118.

VCD file attached (rename from vt.vcd.txt to vt.vcd).

[vt.vcd.txt](https://github.com/user-attachments/files/20700600/vt.vcd.txt)

tbybell, 2025-06-12:
That file works fine for me, too.

tbybell, 2025-06-12:
OK, you're right...time for new glasses.  I thought I was looking at "Time:"

`if(gch==' ') /* in case time is of format "100 ps" with a space */`

...there needs to be some separator in there to ensure the 'S' isn't picked up as a time value.

rfuest, 2025-06-14:
I've implemented a fix for this issue in the development branch. It uses a regex to extract the numbers from the text, which, in my opinion, was easier to implement and also makes it easier to maintain in the future. By extracting the numbers at load time and using the tagging system of `GtkTextBuffer` I've also been able to highlight the clickable numbers visually and change the cursor on hover:

[image]

The changes are limited to `logfile.c`, so this might be a candidate for backporting to LTS. I think only the `handle_number` function would need to be changed.

tbybell, 2025-06-14:
Ok, thanks, I'll look into backporting that into LTS.  I'm always interested in usability improvements.

I need to look at the text buffer capabilities.  The regex search is interesting because one possibility I can imagine is being able to tie the logfile buffer to the pattern search, so for example, tagged transactions can lay down blue markers without having to go into the RTL.

closed 2025-06-14 by tbybell