gtkwave/gtkwave #446: Crash when trying to draw a `real` value

filmil, 2025-06-21:
I have a VCD file generated by Vivado, which contains a `real` type constant named `sim`.

Trying to display this value crashes GTKWave 3.3.104, but also crashes the top-of-tree as of today.

GDB says:

```
Thread 1 "gtkwave" received signal SIGSEGV, Segmentation fault.                                                                     
0x00005555555865e8 in vtype (t=t@entry=0x555555d725d0, vec=0x3 <error: Cannot access memory at address 0x3>) at ../src/baseconvert.c:1086
1086        pch = ch = cvt_table[(unsigned char)vec[0]];                                                                            
(gdb) bt                                                                                                                            
#0  0x00005555555865e8 in vtype (t=t@entry=0x555555d725d0, vec=0x3 <error: Cannot access memory at address 0x3>) at ../src/baseconvert.c:1086
#1  0x0000555555598ece in draw_hptr_trace_vector (self=0x5555556a96c8, which=<optimized out>, h=0x555555e68020, t=0x555555d725d0, colors=0x5555556a96c8, cr=0x555555df7940) at ../src/gw-wave-view-traces.c:1191
#2  gw_wave_view_render_traces (self=self@entry=0x5555556fd4b0, cr=cr@entry=0x555555df7940) at ../src/gw-wave-view-traces.c:194
#3  0x000055555559bf44 in gw_wave_view_draw (widget=<optimized out>, cr=0x555555dc9e00) at ../src/gw-wave-view.c:649
#4  0x00007ffff779fb74 in  () at /lib/x86_64-linux-gnu/libgtk-3.so.0   
#5  0x00007ffff7580835 in gtk_container_propagate_draw () at /lib/x86_64-linux-gnu/libgtk-3.so.0
#6  0x00007ffff758093d in  () at /lib/x86_64-linux-gnu/libgtk-3.so.0   
#7  0x00007ffff7608821 in  () at /lib/x86_64-linux-gnu/libgtk-3.so.0                  
#8  0x00007ffff75782f1 in  () at /lib/x86_64-linux-gnu/libgtk-3.so.0        
#9  0x00007ffff758c862 in  () at /lib/x86_64-linux-gnu/libgtk-3.so.0                                                        
#10 0x00007ffff76094d5 in  () at /lib/x86_64-linux-gnu/libgtk-3.so.0                                                                                                                                                                                                    
#11 0x00007ffff779fb74 in  () at /lib/x86_64-linux-gnu/libgtk-3.so.0                                                                                                                                                                                                    
(... rest omitted ....)
```

Examining the code that crashes:

```
1189            /* draw trans */
1190            if (!(h->flags & (GW_HIST_ENT_FLAG_REAL | GW_HIST_ENT_FLAG_STRING))) {
1191                type = vtype(t, h->v.h_vector);
1192            } else {
1193                /* s\000 ID is special "z" case */
1194                type = GW_BIT_COUNT;(gdb) p *t
$2 = {t_next = 0x0, t_prev = 0x0, t_grp = 0x0, t_match = 0x0, name = 0x555555a5d39d "sim", name_full = 0x0, asciivalue = 0x0, transaction_args = 0x0, asciitime = 0, shift = 0, shift_drag = 0, d_minval = 0, d_maxval = 0, d_num_ext = 0, n = {nd = 0x55555594ee00, 
    vec = 0x55555594ee00}, flags = 40, cached_flags = 0, f_filter = 0, p_filter = 0, t_filter = 0, e_filter = 0, t_color = 0, t_fpdecshift = 0 '\000', is_cursor = 0, is_alias = 0, vector = 0, shift_drag_valid = 0, interactive_vector_needs_regeneration = 0, 
  minmax_valid = 0, is_sort_group = 0, t_filter_converted = 0}
(gdb) p *h
$3 = {next = 0x555555e68040, v = {h_val = 3 '\003', h_vector = 0x3 <error: Cannot access memory at address 0x3>, h_double = 1.4821969375237396e-323}, time = 0, flags = 0 '\000'}
(gdb) 
```

The code seems to use the value of `h->v.h_vector` which is not initialized properly, based on the fact that `p->flags == 0`.

The value of `h->v.h_vector` is passed into `vtype`, which attempts to read from it and fails, since the address is `0x3`, which is not valid.

filmil, 2025-06-21:
Here's the offending VCD file: [sim.vcd.gz](https://github.com/user-attachments/files/20850408/sim.vcd.gz)

filmil, 2025-06-21:
The problem signal is `//tb/uut0/cpu0/sim`.

tbybell, 2025-06-22:
Right.  It shouldn't crash on input.  I'm looking at a fix starting with the VCD recoder, at least to continue parsing and print a warning.

AE is the Application Engineer, which is typically the point of contact from customer into the engineering department of a company when filing a bug report.

From the sound of it, no it's not a real: that's what some other simulators in the past have aliased as a size to get real parameters in the VCD because real parameters in VCD files didn't exist in the past (and they still might not).  If they're dumping booleans, the size needs to be *1*, not a *0*.  That is the problem in the VCD file and is a bug in their simulator, so disregard any observations involving "real" in my previous comment.

filmil, 2025-06-22:
> Right. It shouldn't crash on input. I'm looking at a fix starting with the VCD recoder, at least to continue parsing and print a warning.

Thank you. Some way to continuing to work is preferable to terminating the session.

> AE is the Application Engineer, which is typically the point of contact from customer into the engineering department of a company when filing a bug report.

I'm a cheapskate, I have no point of contact to report this to. :/

tbybell, 2025-06-22:
OK, then I'll "Make It Work(tm)."

tbybell, 2025-06-22:
d3de62e..47e2bd8  lts -> lts

This is only LTS, not master.  The length 0 fix is marked with if(1) with the old code in the else branch.  I'll break out Questa to see if this is still an issue.  If so, I'm going to have to gate the old MTI behavior back in by searching for ModelSim or Questa in the $comment block.  Whatever the case, this should get you up and running, thanks.

```
3.0.25  10apr07 System Verilog with MTI fix for VCD declarations of form
                $var reg 64 >w #implicit-var###VarElem:ram_di[0.0] [63:0] $end
                ...debussy implicitly escapes the varname during loading so
                gtkwave does it too now for all VCD loaders.
                More System Verilog with MTI fixes: VCD (parameter) vars of
                length zero are representative of reals with MTI: they don't
                use real_parameter like Riviera does.
                Updated vcd_saver.c to handle saving these implicit-var
                facilities correctly.  Likewise updated lxt-write.c,
                lxt2_write.c, and vzt_write.c to handle correct bracket
                stripping for the [0.0] semantics.
                Modified dynamic resize routine so large escaped names don't
                cause issues with the signal window crowding out the
                wave window.
```

tbybell, 2025-06-22:
I tried QuestaSim this afternoon, and it still has the old behavior, so I had to make the old behavior conditional on Questa/ModelSim, so the if(1) marking of code turned out to be useful.

I also noticed the MTI behavior was missing from vcd2fst so I added it where the variable declaration is changed to a real_parameter one.

47e2bd8..deb3515  lts -> lts

As a testcase for Ralf and anyone else who would be interested (e.g., Surfer developers), 
[des.vcd.gz](https://github.com/user-attachments/files/20854534/des.vcd.gz)

Relevant Lines of note are the following, but I included des.vcd as it came straight out of the simulator:
```
$version
	QuestaSim Version 10.7d_2-QCF
$end
```
`$var parameter 0 " rr $end`
`r3.14159 "`

Legacy name version strings will appear like this:
```
$version
	ModelSim Version 5.2a
$end
```

filmil, 2025-06-24:
Thanks for checking. I suppose that's what I get for using VCD from Vivado. :/

tbybell, 2025-06-24:
Note that if you're using the VCD files directly from Vivado, they should now work fine with the latest changes (although it will show messages/warnings during loading).  If you're converting those VCDs to FST to make it more compact and faster, some parameters which should be real will be missing values and appear as X because the input VCD file is not sane.

It might be a matter of component vs entity instantiation.  Does Vivado write VCD directly from an invocation of $dumpfile/$dumpvars on the Verilog side, or is there a separate VCD export from an internal format?  I still might experiment with xsim, but I didn't want to wait for 20GB to download this evening.

filmil, 2025-06-25:
The VCD file I sent is what comes out of `xsim` when you add a VCD generation option. How it produces the file is opaque to me.

I'll try the instantiation variations.